umigv/MkII #20: Research - Best way to deliver cv results

apvarghese19, 2021-11-08:
Issue: We need to figure out the best way to deliver the cost map

Two Options:
1. Generate a 2D cost map of probabilities of a white line/ pothole
2. Generate a 3D point map and create walls in place of the white lines to combine with 3D map from sensors

Task: Write a comment on which approach you think would be better and why

vishwap2, 2021-11-21:
For using a 2D Cost Map, there already is an API readily available in the ROS system and in MathWorks called an Occupancy Object, which will give you the probability of the occupancy of that cell. So, if the probability is close to 0, it's not occupied and is obstacle free, and vice versa if it's closer to 1. For using a 3D Point Map, there is an infrastructure available in Rockworks which takes in X, Y, and Z coordinates to produce the map, so this would place walls in the map to make sure robot avoids running into them. Based on these two options, I think using the 2D Point Map would be the least computationally expensive option since it doesn't have to render a whole display in a 3D plane, rather just give probabilities of coordinates' occupancies. However, this may be limited in terms of what data can be interpreted from here and may lead to some incorrect decisions being made.

dswelch, 2021-11-29:
I believe that using a 3d point map will be the best way to go forward. I don't believe that creating the 3d point map will be that much more difficult than creating a bird's eye view 2d cost map, and we know that sensors will need to convert it to a 3d map eventually anyways. I think that the big question will be how much more intensive the 3d map will be, but I believe that the 3d map is the more efficient way to go.

eliasort, 2021-11-30:
A 3d cost map has more depth to it and requires more complexity. However, since the 2d map has many of the same elements as a 3d map it can also work. Since a 2d cost map will be just as applicable I believe that is what we should focus on since it will probably be not as complicated. We will have to convert it into a 3d map when it is combined with the sensors but it will be more efficient this way.

Garadan1, 2022-01-30:
Thanks for your good work on this, everyone. Moving this to Review/QA and assigning to @apvarghese19, @hvak, and @Garadan1 for review.

closed 2022-02-20 by hvak