umigv/MkII #41: Basic Lane Detection

dswelch, 2022-01-30:
Basic lane detection with still images

dswelch, 2022-02-20:
Created kaylaLaneDetectionCopy.py, which is a copy of kaylaLaneTest.py except edited so that it works with laneDetection.py. Basically just put everything into a class so that different functions could be called. End goal being that the lane detection can be used on an svo file (video) rather than a single frame. Running into an error though

kaylalee1116, 2022-02-20:
Created WhiteDetection.py and edited KaylaLaneTest.py. whiteDetection.py is a work in progress but I'm thinking about maybe only detecting white and then isolating based off of coolor. In KaylaLaneTest.py I am trying to figure out why the blobs show up as a different amount every time.

dswelch, 2022-03-06:
Worked on laneDetection.py. Running into problems with getting the kaylaLaneTest code to a video so tried running it with ADS detection, and other than the recording with 01 in it I was not able to get it to work with any others.

closed 2022-08-22 by dswelch