umigv/MkII #48: Edge detection

apvarghese19, 2022-02-06:
Find best implementation of detecting lanes/potholes using edge detection

dswelch, 2022-02-10:
Used https://www.kdnuggets.com/2017/07/road-lane-line-detection-using-computer-vision-models.html in order to use Canny edge detection on a single frame. We then started trying to apply this continuously to a svo file/video.

mmiracky, 2022-02-10:
Tried using a white mask before and after grayscaling an image to get only white lines. Issue: Tuning the threshold manually is inconsistent. We ended up dropping this idea.

ryanhou28, 2022-02-10:
Tested techniques from https://medium.com/analytics-vidhya/building-a-lane-detection-system-f7a727c6694 to find edges and lines after manually isolating the area of the image that contains the lanes. The perspective transform and line detection ended up working quite well.

mmiracky, 2022-02-10:
Tried using https://data-flair.training/blogs/road-lane-line-detection/ to detect white/yellow lines. There were lots of errors that even people in the comments were talking about being unable to fix. The main error was that the code tries to loop over a None. There were also other errors in the code like using actual greek letters instead of alphanumeric.

dswelch, 2022-02-10:
We were able to apply the edge detection and hough transform from ADSDetection.py onto a video, and then using VideoWriter from python we could export that video into an avi file in order to see it as a video.

ryanhou28, 2022-02-10:
Finished a working implementation of circle/blob detection.

ryanhou28, 2022-02-13:
Worked on fixing blob detection for perspective-transformed images. Used Gaussian blurring and color segmentation to get better shapes for detection. Also tried various parameters for the blob detector. Still need to find a way to detect blobs by area.

closed 2022-08-22 by dswelch