umigv/MkII #60: Perspective transform

dswelch, 2022-03-13:


dswelch, 2022-03-13:
Get a message from a ROS topic, that contains the image from the zed camera (which is already distorted). Then we have perspective_transform node that would create the correct points on initialization, then use those points in order to do a perspective transform. It will then have a publisher that will post that topic in order for another node to use that and create the occupancy grid.

apvarghese19, 2022-03-17:
Need to figure out how to import cv_bridge, try to configure python3 with pros melodic

dswelch, 2022-03-20:
Finished perspective transform ROS node, started organizing files within catkin workspace and cvStack. Rtabmap is no longer working for the occupancy map. Odometry data seems to be the problem

dswelch, 2022-03-27:
Got all ROS nodes working in order to visualize our lane detection algorithm, including the perspective transform. Still need to figure out how to convert to occupancy grid.

apvarghese19, 2022-03-31:
Using dst values in order to only look at selected portion of the transformed image and ignore the black space. We are doing this by publishing these values to the next step.

dswelch, 2022-04-05:
We have a document that gives the steps to get all of the ROS things up and running in the Computer Vision drive. We now give the dst_quad values properly, and when we run the hough transform the only lines that are picked up from the edge of the image is at the very top corners, which should be fixed fairly easily. Now we need to work on lane detection and pothole detection.

apvarghese19, 2022-04-07:
Fixed the previous issue, we were using the wrong points from the dst and masking it weirdly but it is working now. We looked into more lane detection techniques, need to still work on it

closed 2022-08-22 by dswelch