umigv/MkII #61: Occupancy grid

apvarghese19, 2022-03-13:
Listen to the left transformed and right transformed topics and finds and edits an occupancy grid that we will publish to a topic

dswelch, 2022-05-02:
Put some possible code in ADSDetection.py for creating an occupancy grid. Unsure of height, width, and resolution necessary to combine with sensors' occupancy grid as of right now, but this code looks like it could create the necessary grid. Also put two helpful links at the bottom of CV winter 22 plan.

dswelch, 2022-05-05:
The occupancy grid is now being published under the topic /cv/laneMapping/ogrid. Unknown values are -1, where we detect lanes are 100, and everywhere else is 0. We should have a confidence level instead of just 0 or 100 eventually. We need to make sure that the potholes are being included within it as well. Edge detection is a big problem right now in the svo files we have, and it is causing us to miss some lanes in the svo. Also, less than 1 hertz so we need to do a lot of opitimizing.

apvarghese19, 2022-05-06:
We found that creating the occupancy grid is the task that slows everything down. Need to try different solutions in order to optimize. Another task needed is to make sure the occupancy grid is able to combine with cartographer data.

dswelch, 2022-05-13:
Using C++ for the occupancy grid did not help performance. We need to figure out where we can optimize. Some thoughts are identifying that the unknown pixels will always be at the same spot so we don't have to set them every time. Also, we need to look into optimizing the perspective transform and definitely the lane detection because the more lanes we detect the slower it goes by a noticeable margin.

closed 2022-08-22 by dswelch