umigv/MkII #66: Lane Detection

dswelch, 2022-04-10:
Considering using color thresholding to detect lane lines and potholes. Either thresholding immediately in order to isolate white (or close to white) parts of the image which should be things we want to avoid no matter what, or we will take the white thresholded image and compare to our edge detection algorithm to create a probability map that records white lines. Also need to make sure that we can adjust our threshold easily the day of the competition to make sure that we actually detect the correct colors. Further testing is necessary.

apvarghese19, 2022-05-06:
Fixed Lane Detection so that more pixelated/hard to see white lines are picked up. Edited the thresholding in Canny edge detection as well as thresholded the image after displaying the potholes in order to avoid catching non-white potholes.

dswelch, 2022-05-31:
Now have everything working again on the Jetson, working on getting cuda working so we do things on the gpu instead of cpu. Running into an error for createGaussianFilter, also the functions that draw onto the image can be tricky because I cannot find cuda functions that draw on images (but I might just be missing it) so that would be a lot of going back and forth.

closed 2022-08-22 by dswelch